PancakeBot/PancakePainter #17: M1 wrap update

techninja, 2015-07-30:
_This issue serves as placeholder and discussion point for general wrap for the **M1 milestone**. It should be the last ticket closed for the milestone, and have attached in comment the final build. Any specific problems found in the build should be taken up in the feature request issue it pertains to, otherwise general comments can be made here. The milestone is only complete when @Slimmy82 closes this issue._

techninja, 2015-07-31:
Ok, a pretty productive week. All tickets down, and the app is building great.
Wrap build check here: [PancakeCreator_m1_0.1.0.zip](http://tn42.com/PancakeCreator_m1_0.1.0.zip)

**Notes: **
* Window will remember its size/position on the screen
* Not much going on yet. This is the shell where the application will live starting in full next week.
* The icon was a quick fake-up as it needed something a little more to be visible. Again, likely a placeholder.
* Should be in great shape to start getting real, with all libraries in place and everything building smoothly.

Slimmy82, 2015-08-03:
2 issues
1)Menu names need changing
FILE
-New
-Open
-Save
-Save as
-Export to Gcode (maybe Export to Pancake)
[image]
2) When you close the window (red button top left of window) it closes and there is no way to re-open it without restarting APP

techninja, 2015-08-05:
* App now closes on window close for simplification
* The problem with the menu translation was that ic couldn't find the translation files only when the app was built! That's fixed now as well. Closing this issue.

closed 2015-08-05 by techninja