PancakeBot/PancakePainter #19: M3 wrap update

techninja, 2015-07-30:
_This issue serves as placeholder and discussion point for general wrap for the **M3 milestone**. It should be the last ticket closed for the milestone, and have attached in comment the final build. Any specific problems found in the build should be taken up in the feature request issue it pertains to, otherwise general comments can be made here. The milestone is only complete when @Slimmy82 closes this issue._

techninja, 2015-08-15:
Here's the basics for this weeks update: [PancakeCreator_m3_0.6.0](http://tn42.com/PancakeCreator_m3_0.6.0.zip)
* Gcode render has gotten at least 2 overhauls, now includes: 
  * #21 Feedrate output
  * #10 Proper timing, path splitting & Pump line endings
  * Tons of improvements for code quality and setup
  * GCODE is now fully annotated, declaring each path segment and its intentions.
  * #13 Beta Color separation timing
* #12 New "tool" for color selection. Didn't worry too much about the looks for now, but it seems to work. Click to select the color you want to draw with, and click a color to change the currently selected line.
* Bonus for this week: #24, closing paths.
  * Any path you're currently drawing can be brought back around to its start and be joined. This seems to have introduced some oddities in path simplification with freehand paths, but It isn't reliably bad enough for me to fix with clarity.
  * I added an obvious green circle that pops up when it will snap closed, allowing the polygonal tool close with one click instead of double or otherwise. This can be changed in size or sensitivity easily.
* #11 Full bot simulator! This simulates basically any GCode you can throw at it and moves/extrudes batter in a realistic way while slowly cooking the batter to change its color. This will likely find a home in its own repo, but likely after launch.

Slimmy82, 2015-08-15:
Issue with the free hand tool. When you draw free and and release it smooths way too much. I tried to draw a circle. upon closing the circle and releasing it turned into a square. I can email you video if you like

techninja, 2015-08-15:
Yea, I mentioned that. I'll tweak it a bit and re-upload. Feel free and provide a video if you can. It's quite tricky.

Slimmy82, 2015-08-15:
[image]
The above is just before linking and releasing
The image below is result
[image]

techninja, 2015-08-15:
Aha, found the problem: I added an additional smooth operation right after the initial simplify. The problem is that they're not very compatible, meaning that the output from simplify tries hard to be very much like the line you drew (which it does!), but then the smooth operation tries to even out the curves for every point, which the simplify operation was not trying to do.

The reason I added the other smooth operation, is because any time you touch a point it's going to run the smooth operation. Which gets us back to some of the issue around #23. Do we run the smooth but wreck the shape so users can easily shift the points? Or do we not run the smooth at all? or somewhere in between?

techninja, 2015-08-15:
I'm going to upload a version that has ZERO smoothing and you'll have to compare it to the current version and see if we want to scrap smoothing.

techninja, 2015-08-15:
Here it is: [PancakeCreatorNoSmooth_m3_0.6.0.zip](http://tn42.com/PancakeCreatornosmooth_m3_0.6.0.zip).

This removes any and all path smoothing on nodes or during line creation. Freehand paths are much happier, but moving nodes around isn't quite as useful or intuitive. Possibly a reasonable tradeoff.

closed 2015-08-17 by techninja