PancakeBot/PancakePainter #20: M4 launch wrap update

techninja, 2015-07-30:
_This issue serves as placeholder and discussion point for general wrap for the **M4 milestone**. It should be the last ticket closed for the milestone, and have attached in comment the final build. Any specific problems found in the build should be taken up in the feature request issue it pertains to, otherwise general comments can be made here. The milestone is only complete when @Slimmy82 closes this issue._

techninja, 2015-08-20:
There are so many things stuffed into this last milestone that directly effect user interface and experience that I'm releasing it early so we can ensure everything gets TESTED. Lots and _**lots**_ of testing is needed for this release. I've already had my kids give it a go and they've found some bugs. Specific to this release 0.9.0 > [PancakeCreator_m4_0.9.0MAC](http://tn42.com/PancakeCreator_m4_0.9.0MAC.zip) | [PancakeCreator_m4_0.9.0WIN](http://tn42.com/PancakeCreator_m4_0.9.0WIN.zip):
 * #14, #15 for file input/output and current file management with edit detection and warning dialogs on new, close, open, etc.
 * #16 for image import! This is giant but is meant to feel like a "silent" addition. The transparency amount is likely up for debate, but everything else works pretty well as expected.
* #13 & #25 are still open and need to be managed in 0.9.1 or later before 1.0 release. Shims have been put in, but real discussion is needed before this is finalized and has been tested with a real PancakeBot.

This early release is really about UI testing and verification. Everything in it should work (apart from the color separation work), so if anything at all acts funky, now is the time to call it out.

Slimmy82, 2015-08-20:
Great, I will download and play around with it. I will also get a few people to test and play with it.

techninja, 2015-09-05:
OK, after a few more weeks of work, I've implemented the very first required parts for fill creation and rendering.  Specific to this release 0.9.5 >  [PancakeCreator_0.9.0MAC](http://tn42.com/PancakeCreator_0.9.5MAC.zip) | [PancakeCreator_0.9.5WIN](http://tn42.com/PancakeCreator_0.9.5WIN.zip):
 * #28 - Horizontal mirroring of GCODE by default, with an additional menu item for non-mirrored output
 * #31 - Full design implemented 
 * #29 & #30 - Paint bucket style fill and GCODE rendering of filled areas. This is huge and likely because of complexity will have some bugs. Needs to be tested, tested, tested!

techninja, 2015-09-09:
I've done some general tweaks to fill for #33, fixed a GCODE render issue with very small line lengths (Which may or may not help as we're turning the pump on and off fairly quickly), at least every path should turn the pump off afterwards.

[PancakeCreator_0.9.6MAC](http://tn42.com/PancakeCreator_0.9.6MAC.zip)

techninja, 2015-10-08:
All previous version have been invalidated/removed, this is now the final Release Candidate. Once this gets proofed & tested, I'll initiate it as a full release on the main page. [PancakePainterMAC-v1.0.0-RC1](http://tn42.com/PancakePainterMAC-v1.0.0-RC1.zip)

Public change list (from original email):
   * Griddle dimensions now perfectly match the bot understood GCODE units (half mms), and preview perfectly matches the presto griddle. 
   * Name Changed, Logo Changed & Icon Changed.
   * Settings window: Lists all the tweakable settings for gcode rendering. Saves immediately, resettable.
   * Fill output improved, added wait cursor for long fills, fixed "can't fill this thing" problem, and added an error catch to it. This was also killing some saved files from being re-opened, so double win.
   * Drawing is now left mouse only, allows for right mouse click end polygonal path. Double click is somewhat broken as the first click goes to the red dotted, then the second click goes to the new path. Double clicking on the path you just made will end it though.
   * All file warning dialogs are now native, with explanatory button text
   * Export for print has a nice "please wait" window now, for complicated exports
   * Logo now fits on the screen under a certain width
   * Added version number next to the logo, and GCODE notes when it was made by what version, and with what settings (good for tech support)

techninja, 2015-10-20:
All previous version have been invalidated/removed, this is now the final Release Candidate. Once this gets proofed, I'll initiate it as a full release on the main page. [PancakePainterMAC-v1.0.0-RC2](http://tn42.com/PancakePainterMAC-v1.0.0-RC2.zip)

#### Change list:
* 96405ac: Only add a new path segment node if the path is already selected
* New App Icon from designer (looks good!)
* #35: Fix all bad paths effected by paperjs issue 801 on fill click
* #34: Update build to Electron latest 0.34.0 released 3 days ago.
* #38 & #36: Redo end of path handling code to be more inclusive, and ignore extra work on very short paths.
* 5e494259: Catch and remove stackblur alert boxes that would occasionally kill app at startup with silly messages.

techninja, 2015-10-25:
The RC2 release above has been officially released as 1.0.0-Beta1, so M4 extended Is complete.

closed 2015-10-25 by techninja