PancakeBot/PancakePainter #22: Good Example GCode

techninja, 2015-08-10:
Given the "bull" GCode example to base my simulation code off, I've done a quick analysis on the movement data:
```
G1 Move:
X: {min: 216.876, max: 623.116}
Y: {min: 27, max: 393}
```

Our draw area should look something like this:
```
 [485, 0]                 [42, 0]
    +-----------------------+    
    |                       |    
    |                       |    
    |                       |    
    |                       |    
    +-----------------------+    
[485, 210]              [42, 210]
```
And I'm assuming the design diagram I got shows the "front" of the bot here to the bottom, meaning the "top right" of the bot is the home position and `[0, 0]`.

This means our "perfect" GCode is upside down, and flies way off the left edge and bottom by nearly 200mm?

techninja, 2015-08-10:
Here's the bull Gcode source for reference:

```text
G21 ; set units to millimeters
M107 ;/Air pump off
M84 S1 ;/pause 1s
G28 ; home all axes
G1 F6600
M106 ;/Air pump on
M84 S2 ;/pause 2s
G1 X239.366 Y200.940 ;/Coordinated Movement X Y;
G1 X351.516 Y49.350
G1 X352.636 Y47.910
G1 X355.586 Y44.340
G1 X357.516 Y42.200
G1 X359.616 Y40.230
G1 X361.866 Y38.450
G1 X364.276 Y36.860
G1 X366.796 Y35.490
G1 X369.436 Y34.330
G1 X372.156 Y33.400
G1 X377.466 Y31.830
G1 X379.236 Y31.350
G1 X384.616 Y30.070
G1 X386.396 Y29.700
G1 X391.836 Y28.710
G1 X393.686 Y28.430
G1 X399.176 Y27.740
G1 X401.016 Y27.550
G1 X406.536 Y27.160
G1 X408.446 Y27.080
G1 X413.976 Y27.000
G1 X426.016 Y27.000
G1 X431.546 Y27.080
G1 X433.456 Y27.160
G1 X438.976 Y27.550
G1 X440.816 Y27.740
G1 X446.306 Y28.430
G1 X448.156 Y28.710
G1 X453.596 Y29.700
G1 X455.376 Y30.070
G1 X460.756 Y31.350
G1 X462.526 Y31.830
G1 X467.836 Y33.400
G1 X470.566 Y34.330
G1 X473.206 Y35.490
G1 X475.746 Y36.880
G1 X478.146 Y38.470
G1 X480.416 Y40.260
G1 X482.516 Y42.240
G1 X484.446 Y44.390
G1 X487.386 Y47.960
G1 X488.466 Y49.350
G1 X600.626 Y200.940
G1 X602.476 Y203.710
G1 X604.046 Y206.640
G1 X605.316 Y209.720
G1 X606.276 Y212.910
G1 X620.426 Y269.490
G1 X620.826 Y271.320
G1 X621.486 Y274.810
G1 X621.906 Y277.430
G1 X622.286 Y280.260
G1 X622.446 Y281.750
G1 X622.786 Y285.550
G1 X622.926 Y287.560
G1 X623.096 Y292.240
G1 X623.116 Y293.610
G1 X623.096 Y297.700
G1 X623.056 Y299.100
[... 943 lines cut ...]
G1 X347.898 Y172.264
G1 X347.063 Y170.724
G1 X348.000 Y170.255
G1 X349.365 Y169.728
G1 X350.719 Y169.370
G1 X351.713 Y169.192
G1 X352.442 Y169.118
G1 X353.558 Y169.074
G1 X354.315 Y169.103
G1 X355.474 Y169.224
G1 X356.238 Y169.364
G1 X358.139 Y169.916
G1 X359.913 Y170.732
G1 X361.562 Y171.806
G1 X363.032 Y173.110
G1 X364.295 Y174.616
G1 X365.323 Y176.293
G1 X366.091 Y178.098
G1 X366.744 Y180.608
M107
M84 S1
G28 X0 Y0 ; /home X axis
M84 ; /disable motors
; filament used = 32063.6mm (629.6cm3)
```

techninja, 2015-08-10:
Also interesting is this passage in the middle for the switch from outline to the head of the bull:
```
G1 X235.956 Y206.640
G1 X237.516 Y203.710
M107 ; Pump Off
G1 X239.116 Y201.314 ; Outline ^^
M106 ; Pump On
M84 S2
G1 X369.641 Y95.103 ; Bull Head (from here down)
G1 X370.855 Y90.840
```
The pump turns off right before the last movement to complete the outline, that's great, and expected. But right before it moves to the inside image, it turns the pump back on (basically immediately), waits a full 2 seconds then draws a big messy line from the outline to the head. Does that seem right? :P

Slimmy82, 2015-08-10:
Not sure about this. We will need to wait till we have a machine to test on

techninja, 2015-08-10:
The creation of a simulator as described in #11 is meant to alleviate this, at least temporarily. Initial work (with heavily adjusted dimensions, see first comment) has resulted in the following output for the untouched bull:
[image]

I then adjusted the position of the pump on and wait to be **_after_** the move to the next point (as expected), and got this:
[image]

Either something very weird is going on with how these instructions are meant to work or be executed, or the given bull GCode is wrong. Will test with my other example.

techninja, 2015-08-13:
Testing with Miguel's Star Wars ATAT GCode file doesn't yield much better in the realm of accuracy to the draw limits, and it uses a different format of G1 (G00?). Regardless, I've built the simulator with universal remap, allowing completely custom limits to be set for rendering, so here it is!

[image]

This properly approximates batter pump warm up and turn off lag, as can be seen in the occasional trail off. I suspect the problem seen in this gcode isn't nearly as bad on a real bot... but of course I've not yet seen one. The great thing is the warm up timing on this is perfect (as it uses fractions of a second for timing, something I believe is going to be 100% required for the firmware).

I'm going to continue to move between the two GCODE examples here, but I hope to see a new official GCODE from the guys in China after tooling is done that we can base our standard on.

Slimmy82, 2015-08-13:
More than likely China are clueless about Gcode. Rather, we may want to send them a gcode file and have them test it to see the results.

techninja, 2015-08-13:
That's certainly possible. Of course, didn't they give us the "bull" GCODE? I'd say it's a fine start to say "tell us how this prints", because according to what I've seen from their GCODE limits, it will come out terribly and it will help them understand we need something different, otherwise it's going to be quite a surprise when we get our first bot.

techninja, 2015-08-21:
Ok, Miguel emailed me a template of how dimensions are meant to be:

[image]

This is apparently not in MM, unless the width of the griddle is nearly 4 feet wide :straight_ruler: :stuck_out_tongue_closed_eyes: 

This expands the "printable" area by a notable amount, and doesn't do much to clarify the "top/bottom" issue, so I've asked for clarification and should get more info soon.

techninja, 2015-08-24:
Looks Like we have final confirmation on general GCODE commands! Will make list of remaining todos for this ticket in top comment.

i-make-robots, 2015-09-05:
G0 and G1 are the same, but traditionally G1 was meant to indicate a rapid move between two work areas.

techninja, 2015-09-05:
I believe we've cleared up the GCODE oddities as Miguel sent them what he actually wanted and they adjusted. I think. See latest 0.9.5 GCODE output here: https://github.com/PancakeBot/PancakeCreator/issues/30#issuecomment-137858259

techninja, 2015-10-06:
I'm closing this out as I have something close-ish to real GCODE as used by the bot. There's a few header things I probably don't need to do, but my best guess is I'm just going to have to make the standard as one is unlikely to show up. Neat thing is this is all open source, so if it's wrong, it can be "easily" corrected :)

closed 2015-10-06 by techninja