PancakeBot/PancakePainter #23: Drawing tool

Slimmy82, 2015-08-10:
Currently there are 2 separate drawing tools. 
-Hold mouse button to free hand draw.
-Click move click for polygon tool

Can we join them? for for example you start drawing free hand with the mouse button held down. As you release the mouse button you enter directly into the polygon mode with the first anchor being the release of the free hand tool. While in polyglot mode, if you hold the mouse button down you return to free hand mode with the first nod of free hand connecting to the last node of polygon. This way you can create designs with linked/connected notes using both techniques.

To exit drawing mode one can hit esc key.

techninja, 2015-08-10:
Certainly doable, if possibly a little strange. I'll experiment with how this works out on the local build and report back.

techninja, 2015-08-10:
One major issue in combining them: Currently when dragging nodes/control points in a line, an automatic smooth operation is run that allows path shapes to be modified easily, if it's not run, every new node on a path is polygonal and sharp.

This smooth operation should NOT be run on polygonal sections, as it will turn them into extreme curves, turning a triangle into a blob. I currently have a data attribute on paths created with the polygonal tool to prevent the smoothing for them, but it generally works far better for "pencil" drawn paths to use the smoothing on modification.

If we made the tool like this, because paths could be both smoothed and polygonal, we'd have three choices:
 0. Turn off all path smoothing. (Easy, but removes editing ability)
 0. Smooth everything. (Easy, but messes with the shape of polygonal points)
 0. Only smooth points that aren't polygonal (Hard, but solves the duality, if adding to the complexity)

techninja, 2015-08-11:
Without adding more complexity to this wrinkle, I'm going to wait for your feedback and assign this one back to you. We may need a quick phone discussion to work things out.

Slimmy82, 2015-08-11:
Lets have a quick phone discussion Wednesday (if possible) at a time that works with you.

techninja, 2015-08-11:
Sweet, will give you a call tomorrow morning my time.

techninja, 2015-08-13:
After discussion, we're going to drop this feature request for adding a bit more complexity than we're looking for.

closed 2015-08-13 by techninja