PancakeBot/PancakePainter #26: UI Design Sample

Slimmy82, 2015-08-17:
Here is a sample UI design. 

-The griddle is larger to fill up the horizontal space
-Icons centered above the griddle.
    -The color pallet for shades needs to be adjusted to match current
    -A red box can be placed around the icon showing that it is selected, similar to shade selection (open         to other suggestions
-Logo underneath

Files can be found here
https://www.dropbox.com/sh/ycwrgf1sopxb3yy/AAAfaAAloeLcdEAZiMgIBTRWa?dl=0
[image]

techninja, 2015-08-18:
I hazard to note that maximized/fullscreen for a 16x9 screen (with taskbar, etc), it's our horizontal space that we have more of:
[image]

Note that this is entirely dependent on the resolution and size of the screen. Size of UI elements like the logo and the toolbar (along with the left/right margin) are set absolutely in pixels, so as the size of your screen goes up, these elements are far smaller in relation. 

Here's at around half screen windowed on a 1920x1080 (full HD) display:
[image]

And here's maximized on a 4k UHD display running at ~100DPI (High res macs run around same resolution, but > 300dpi):
[image]

Slimmy82, 2015-08-18:
Ah ok, so is it better to have the toolbar on the left instead of the top?

techninja, 2015-08-18:
Well.. I think it's a bit up in the air. You should likely get a chance to play around with this before you make a decision, I'll see if I can't get you a build out by tomorrow mid-day.

It personally does not feel "bad" to me, but it may feel awkward to others. And the benefit of a side mounted toolbar decreases considerably from an already small amount the larger the screen is. Shrinking/nixing the logo may get us more, but it pulls the canvas away from center horizontally. These are commonly not trivially solved issues, and will definitely take some tweaking to get right.

techninja, 2015-08-19:
@Slimmy82 Alright, this is basically ready for testing: [PancakeCreator_DesignTest](http://tn42.com/PancakeCreator_DesignTest.zip). Lemme know how that feels for various resized states.

Slimmy82, 2015-08-19:
Maybe make the logo on the bottom smaller, other than that I am happy with it for now. Thanks.

techninja, 2015-08-19:
Currently the height of the logo is matched with the toolbar so the griddle remains completely centered. I can cut the height of the logo by half (or maybe 1/3), and then shift the centering down by that much, but I'm not sure it's worth the visual offset. You want me to make you a new build with this so you can see, or should we just keep it where it is?

Slimmy82, 2015-08-19:
It's ok leave it for now. I rather get other things done. We can make it happen for the next build. No rush with this, we may even need to edit some other elements, depending on how the team likes it tomorrow.

techninja, 2015-08-19:
Got. Feel free and show off the design with this build then.

techninja, 2015-08-21:
I'm going to call the work on this done for now. Feel free to open a new issue when you decide on the "final" release design.

closed 2015-08-21 by techninja