PancakeBot/PancakePainter #27: Free hand drawing needs work

Slimmy82, 2015-08-19:
The problem comes when completing the free hand drawing.

Before the line is connects
[image]
After the end is connected with the start. You see a dramatic change
[image]

Can we please also make the auto connecting area tighter? The problem is when you want to zig zag to fill in an object, you come to close to the initial node even though that is not the intention

techninja, 2015-08-19:
**Lines:** When a line is drawn, a line node is added for every mouse movement (usually thousands for a long curvy line), then a simplification function is run on the line to clean up the points down to a reasonable amount, limited by a number that we give as a factor of how "different" the line can be. I suspect the issue is that I'm removing the "end point", closing the path, then simplifying it. I'll experiment with a different order and see if that helps. I will also lower the amount of acceptable offset (though this will make reshaping lines harder as there will be more points, and remember there's no smoothing).

**End Snap:** Currently a line length of 10 "pixels"  is the distance limit for end snap. I can bring it down by half, but that is the same width as the line. Maybe 7?

techninja, 2015-08-19:
Oh, and as far as filling goes, I'm nearly to the point of saying directly to users: "If you have a closed  shape you want filled, just pause the machine and use a second squeeze bottle to fill the shape yourself". It's easily 10x easier than filling in the app by hand.

Slimmy82, 2015-08-19:
hmm, yeah drawing larger shapes is ok, but its when you draw smaller tighter shapes the ending changes dramatically. 

Yes please drop it to 5 and I can test it on the next build. 

I know exactly what you mean, but the idea of this is automation. As soon as manual labor is involved the value of this goes down. Think about restaurants, cafes, and people that are going to want to just select a design, hit print and not need to worry about it till it's done.

techninja, 2015-08-19:
If there's a pre-made design with fill, there's no extra effort by the end user anyways. For the person making it? Definitely, but that's fine as that's a one time tweaking everything till it's right before launch kind of thing.

Slimmy82, 2015-08-19:
Excellent, yeah. the functionality works great. It's just minor tweaking prior to release that will make it awesome. Very happy with what you have done up to now. Cant wait to test results on the printer!

commit 3cea7eb8 (2015-08-19): Adjust simplify thresh and fix freehand drawing of very small objects
fixes #27
closed 2015-08-19 by techninja