PancakeBot/PancakePainter #28: Gcode horizontal mirroring

techninja, 2015-08-24:
@Slimmy82 I know we discussed this feature over the phone, I just wanted to properly document it for clarity.

When a user draws something, the intent is that they _see_ what the pancake will actually look like _when flipped over_. The idea being that the drawing surface is actually on _the other side_ of the batter that is being placed down onto the griddle.

We can either force this as expected default for ongoing exports, _*or*_ provide two export modes:, one defaulting to mirrored, and one with the text "Export unmirrored for printing...", or something to that effect.

Slimmy82, 2015-08-24:
Yes, so if the nose is on the right hand side when drawing, it should be the same after it has been flipped.  So it prints reverse but once flipped it will represent the drawing. Make that the default "export" then have another "export mirrored" 

Does this make sense?

techninja, 2015-08-24:
I _**think**_ so. So the "mirrored" output export is really mirrored-mirrored (or rather, not mirrored from what the user sees and will be mirrored when flipped) 

In that way, what we display to users is a bit of a lie, because what you're drawing is actually what it will look like flipped and on a plate, **NOT** how it will be put down on the griddle (and that's kind of what users are doing in the interface!).

Maybe we need a bit of text under the griddle saying "Draw what your final flipped pancake should look like."....?

Slimmy82, 2015-08-24:
Haha, that just confused me so much. We definitely need an explanation for the user.

techninja, 2015-08-25:
I'm going to leave this last part expected to be included in the final v1.0 release design, and will Implement the menus as you described, with a the second "export mirrored" item, though I expect there to continue to be a bit of confusion about which way things are by default without some either visual explanation (early in the documentation), and/or some well worded description under the griddle canvas :)

closed 2015-09-05 by techninja