PancakeBot/PancakePainter #31: Interface concepts

Slimmy82, 2015-08-29:
Here is a concept for the interface. Is this feasible? If so, I will present it to the team for confirmation.
[image]

techninja, 2015-08-31:
Perfectly feasible. Notes:
* That batter implement is looking more and more like a baby bottle :baby_bottle::baby:  
* Is the color on the tip of the bucket/bottle intended to match the shade selected?
* What's the "selected tool" look like? Rounded rectangle border like last time? Same for the rounded color selection?
  * Perhaps for the large icons, also a background color change?
* It's likely we should try to be consistent with fill direction when it comes to cursors, EG, bottle is pointing left, fill icon should probably be left, not right. This is the one I'm currently using:
  * [image]
  * I especially like the idea that it's like a measuring cup filled with batter :smile: 
* Would the selection cursor be large and white like the icon? Like colors work well on the darkish-grey background, but I have to put a while & black stroke on all the icons anyways :/
* I'm guessing the "dark grey sun" in the image import icon is a flub and I'll be keeping it how it currently looks.
* Looks like we're keeping the griddle vertically aligned absolutely between the bottom and the top, meaning there will be the same amount of margin at the bottom and at the top? Without the logo at the bottom we can also shift down and take more vertical space at most wider widths.

Everything else is straightforward and understood.

Slimmy82, 2015-08-31:
Is this version a little better?
[image]

-Not sure how we should make it look like when the tool is selected. Maybe a rounded corner square background darker than the grey in the bar. 
- I think the arrow tool should be the same as the icon like the others yeah?
-The colors of the icons are not really white they are the same grey as the background of the window. light grey.
-The sun is transparent so the grey is the same as the bar grey behind the icon. 
-Not entirely sure what you mean about margins but go with what you recommend.

techninja, 2015-08-31:
Notes:
* Looking good, definite improvement. Batter depositing fill & line look great, and should work fine.
* I can try both lighter and darker rounded grey backgrounds for tool selection, but it's better to have a final design approved first.
  * This doesn't solve the selection for the shade selection circles.
* Yes, I'll make sure the icons match the cursors for each tool.
* I'll match the "white" to the grey.. though it should be noted it is a very slight radial gradient, I'll just match the outside grey.
* Why the sun is transparent is a bit... odd, but I suppose it helps make it look more like a pancake drawing? I figured anything to help the icon be clear about its intentions as something to import/manage photographs or graphics would be useful for users.
* Still no answer on the pancake shade color match for the fill/line icons/cursors. I'm going to say the UI doesn't really need it, and it's an extra if you really want it.
* I'll keep the margins as they are, matched, and you can play around with it on the next build and see if you like it.

I'm going to continue holding off on implementing any of this till I hear an final OK from you that it's exactly what you (and the design staff etc.) are looking for, then I'll match it exactly and we can get the next build out.

Slimmy82, 2015-08-31:
Great I appreciate the feedback. I will get approval before you spend any time on it.

techninja, 2015-09-01:
Discussion is always good and takes up a _tiny percentage_ of time that implementation takes, it's just how things should be done. :)

Hopefully you've got enough to go on here to make the "final" design. It should be noted that I fully expect the design to possibly go through one more iteration after you get the working app in your hands. A Photoshop mockup is one thing, but actually using the app is quite another.

techninja, 2015-09-01:
Oh hey, almost forgot: Were we still going to have some helper text for users about which direction their pancake is going ala #28 ? Needs to be in the "final" design if so.

Slimmy82, 2015-09-04:
Hi James, 
Can you please cal me when you are free to briefly discuss UI? 6467063382

Slimmy82, 2015-09-04:
[image]

Slimmy82, 2015-09-04:
"The design above will resemble the actual pancake once it has been flipped." does this make sense?

techninja, 2015-09-04:
Sure does. I'll put it in, and if you have a change of heart, it's easy to fix in the translation strings :)

commit 1b5f6958 (2015-09-05): Implement "final" design to resolve #31
closed 2015-09-05 by techninja