PancakeBot/PancakePainter #33: Fill is a little unstable

Slimmy82, 2015-09-08:
https://www.dropbox.com/sh/ycwrgf1sopxb3yy/AAAfaAAloeLcdEAZiMgIBTRWa?dl=0

There is a video in that folder called desktop which is a screen capture of a test with an image.

-very temperamental with smaller fill objects (may need a better fill tip on icon)
-issues with gaps I assume. Possibly close wider gaps to prevent flooding of unwanted areas
-We have to specify that you need to create outlines first before filling in otherwise I don't think you can draw on a fill then fill the new shape over the existing fill. (which makes sense in terms of printing thinking)

techninja, 2015-09-08:
As I discussed in the feature issue #29, there is a latent issue with path reordering as seen in the video @ 1:38 & 1:56. This is not a gap problem so much as trying to re-order all the borders of the fill algorithm can lead to "path inclusions", usually triangles or other shapes on top of non-fill areas. I'll work to improve the quality of the path reordering, but because it's very specific to the shapes involved, I need example project files with the issue.

Also as we both originally discussed, you _**can not**_ draw on top of fills and get what you "expect", because it has not been defined what _should_ happen. Right now, fills are simply recolored when clicked again, meaning that you draw a new line on top of a fill and expect it to do anything different. It should be noted that a dark line drawn on a light fill will print first, then have fill over the top, so that may actually be something users might want.

You're right that it's best to train the users to "think with pancakes" in this. Make all the outlines first, then figure out what should be filled where.

Also: Icon fill position has a hitTest radius, so it won't fill near I think a 4pixel radius on top of existing lines. I can pull this down a bit if need be.

Slimmy82, 2015-09-08:
I think my biggest problem I had was the fact the screen was to small . I should have expanded the window so I had a bigger working space. Using a touch pad probably made it 100x more difficult. 

Lets see what Miguel has to say. I just want him to print it before we do any modifications.

techninja, 2015-09-08:
Of course, will wait for Miguel. I know he has the latest build and he said he'd " try to print something later today" some time over the weekend. I suspect he got busy or ran into an issue. I also suspect he's allergic to the GitHub issue queue as we haven't seen him on here yet :/.

I made my window full screen and it was much easier to make my monkey. Though I'm still having issues with the small fills not quite matching the same shape/size. Simplification allows a modicum of size change, but it's an absolute, meaning that if we don't take the size into account, a big shape will have a small change, but a small size fill with a same small change could be the half of the shape!

Using the monkey face image, I'm going to ensure the fill is as smooth as possible without failing at smaller sizes. It should likely be noted that after you make either fills or strokes the shape can be easily modified to better fit (though without zoom, you're limited by your screen resolution)

techninja, 2015-09-08:
Also, as I test the triple cavity occlusion of the monkey face (2 ear colors and face color, a very tricky test!), I find that depending on where I click to fill, I with lose either one of the ears, both ears, or the face with an improper join. I originally experimented with choosing the best starting point to make the fill, but ended up going without it as it stands as the results were more often fine. I'm going to run with this as a real bug and will try to find a fix ASAP. 

The good news here is if the code doesn't change a bit for this, it's still workable as you can always delete a fill and try again.

techninja, 2015-09-09:
Check the latest release at the bottom of #20 and you should see it perform a little better at big and smaller fills (though I left the off any change of the tool hitTest radius).

[image]

techninja, 2015-10-08:
Fill has been improved immensely, and proven to work on the bot as of recent commits.

closed 2015-10-08 by techninja