PancakeBot/PancakePainter #34: Mouse Icon/curser reverts to default system

Slimmy82, 2015-10-17:
e.g. When you click on the draw tool the icon changes to that icon. when you start drawing it randomly changes to the default system curser. Restarting the application doesn't fix the issue. Restarting resolves it. However it randomly reverts back when you open the app for the 4th or 5th time.

techninja, 2015-10-18:
Official word on this is that if it works at all, and doesn't "sometimes", then it's likely a bug in the Chromium browser version. For the next build I'll up the version to the latest and we should see both general speed improvements and likely this issue disappear.

Slimmy82, 2015-10-18:
Great, lets hope so

techninja, 2018-02-26:
No updates on this for a while, guessing it's fixed.

closed 2018-02-26 by techninja