PancakeBot/PancakePainter #35: Fill did not work

Slimmy82, 2015-10-18:
Created outline then got fill error. Here are the console screen shots
[image]
[image]

techninja, 2015-10-18:
You should be able to simply drag the PBP file to a comment here and it will attach.

This issue is known and actual just got fixed in paper.js! 
https://github.com/paperjs/paper.js/pull/806

Unfortunately, the fix is stuck in their develop branch, so we can't get it till they roll a release, which may take a little time. Till then I'll implement a quick fix to implement the cleaning function before every fill. With your "broken" PBP file I can confirm it works.

Slimmy82, 2015-10-18:
It wont upload here as it is an unrecognized file
sent email to you

techninja, 2015-10-18:
What? Those GitHub Guys. Only acceptable formats?! Sheesh. WIll check email for the file.

techninja, 2015-10-19:
With this one checked off the list, see if you can't get me screenshots of the error console when you can't save a file and we'll be in great shape!

closed 2015-10-19 by techninja

Slimmy82, 2015-10-20:
This issue occurred again. I understand I don't have the new build with your fix, but thought it would be good to share a screenshot

[image]

commit 6be602f9 (2015-10-19): Add Path Cleaner function before fills are run, fixes #35