PancakeBot/PancakePainter #36: Dispenser did not shut off

Slimmy82, 2015-10-19:
We printed a spaceman and in the application it looked great with fills. When we started printing just after the outline finished and the dispenser was supposed to reset and wait before filling the code didn't add the shut off. It continued to dispense batter. Maybe it didn't add a stop in the g-code

I have the gcode file and the pcb file if you need to look at it.

techninja, 2015-10-20:
Yep, send those both along in email and will look into it. Hopefully I can match the settings in the GCODE file and duplicate the output and problem so I can ensure it gets fixed.

techninja, 2015-10-20:
I've verified this in the simulator, which is very good as it means I can verify the output, and the bug. Quite a virtual mess on my hands here!

[image]

techninja, 2015-10-20:
Apparently this was an edge case where the last point of a mid length closed polygonal line was reading a length of ZERO on its last point. I changed to reading the actual offset of the segment curve location, and removed some of the other edge case coverage for a single catch for all non-first segments and I think it covers all the bases. It certainly fixes your use case.

[image]

closed 2015-10-20 by techninja

Slimmy82, 2015-10-20:
Great how your simulator can test without the mess!

commit bc6e2995 (2015-10-20): Refactor GCODE path render to better handle short line preshutoff
Fixes #38, and fixes #36