PancakeBot/PancakePainter #37: Weird fill phenomenon

Slimmy82, 2015-10-20:
When filling I encountered this weird phenomenon 

Here is before fill
[image]
here is after fill
[image]

Any suggestions to this issue?

I have the .pcb file and a video screen capture if needed.

techninja, 2015-10-20:
This is a known issue, and described on the readme, with a temporary user fix. Unfortunately there's not enough time available to try and work this out completely. The issue is specifically to do with piecing the fill together from the out of order fill points, and if they're not put back together in the correct order, you're left with end points that are far apart, and you get this strange angle grouping like you see here.

I've done quite a bit of work to alleviate this, and it happens far less frequently than it did, but I've run out of time to hack away at improving it more. The problem is fairly complex as every inside island must be accounted for and could be confused for a similar point distance issue. 

If I stand a chance at fixing the issue, I'll need as many saved example files that it occurs with, as it's somewhat difficult to just make it happen.

Slimmy82, 2015-10-20:
Lets leave this open and I will add more examples as I encounter them.

casperlamboo, 2016-02-24:
I'm working on a project that uses a fill tool similar to the one used in the pancake bot app. We encountered a similar problem with the flood-fill algorithm (we used also used n-dimensional-flood-fill).

To solve this I wrote a small library using the boolean operations of [clipper](https://www.npmjs.com/package/clipper-lib). This library uses vector input to generate fills. It can be found here https://github.com/Doodle3D/fill-path.