PancakeBot/PancakePainter #38: Exporting froze

Slimmy82, 2015-10-20:
Went to export and it froze. Opened the dev tools to see issue 

Possible source of issue
1) The no fill error could have broken the design so it would not export. 
2) To many lines breaks the image.

Could it be the last line in the error codes?

[image]

techninja, 2015-10-20:
Very likely export was broken as part of #35 (all those errors above). Shouldn't ever see errors in the console under regular operation. I have not seen that last one before, referencing "_path". I'll have to look into that. As long as you have this file saved, we can re-test once the next build is made.

techninja, 2015-10-20:
Scratch that, I seem to have been able to recreate this issue, it's not related to #35. Will look into it on a timebox and report back.

techninja, 2015-10-20:
Found it! I got the same "_path" error reliably on the T-Rex after filling it and trying to export. Apparently during export if a line is short enough, when the `preShutoff` is triggered on the first segment, we check for the preshutoff position anyway, even though the position at 0 is always the same.

The `_path` error was happening rarely on some of these zero checks (maybe 1 out of 25). By removing any check to zero, it doesn't look like it happens anymore at all, and the previously problem file now exports and simulation prints like a champ.

commit bc6e2995 (2015-10-20): Refactor GCODE path render to better handle short line preshutoff
Fixes #38, and fixes #36
closed 2015-10-20 by techninja