PancakeBot/PancakePainter #39: Vote for Linux Release Support

techninja, 2015-10-23:
Only Mac and Windows have official releases, though the app works for Linux as well. Post here to cast your vote for adding Linux to the release list.

Al2zone, 2015-12-21:
Ubuntu!  13.04+ is fine.

raster, 2015-12-22:
A Linux release would be awesome!

bdale, 2016-01-27:
We're a Linux-only household... happy to build from source if we need to, but a Debian-compatible package would be awesome!

totto, 2016-01-28:
please - we are also a pure linux household :)

peteruithoven, 2016-02-16:
+1 (Elementary OS)

bdale, 2016-02-17:
FWIW, I have the current code running on my Debian notebook just fine...

raemarks, 2016-06-23:
+1 for Fedora! I would love this to be fully supported on Linux.

TechRunner2, 2016-07-27:
Arch and Ubuntu Packages would be amazing. Even just on the aur.

ikanc, 2016-10-29:
+1 for linux! :+1:

shbert, 2016-10-29:
+1 for linux.. rapberry

magnets99, 2016-12-06:
x64 ubuntu or a raspberry pi build would be great!

drtorq, 2016-12-10:
Here's an article I wrote on running the PancakePainter on Ubuntu.
["Hack the Pancake Stack: Running PancakeBot under Xubuntu Linux" ](http://thenewstack.io/hack-robot-stack-pancakebot-running-xubuntu-linux/)

techninja, 2016-12-24:
[image]

Thanks to your votes _(and hard work from the electron userland community)_, we'll be including Ubuntu (deb) and Redhat (rpm)  x64 install files for alpha release coming before the new year :gift:!

xsuchy, 2017-10-16:
Do you have SPEC file for RPM somewhere available? I can likely submit few enhancements.

techninja, 2018-02-26:
@xsuchy I'm using a tool to build using SPEC file-like definitions from the package.json `releaseInfo` section. I don't have a redhat install to test against, feel free and look into the build system `build/Gruntfile.js` under `electron-installer-redhat`, line 174 😄

closed 2018-02-26 by techninja

xsuchy, 2018-02-26:
I am not fluent in nodejs. Can you give me please hint which arguments pass to electron-installer-redhat to genereate rpm or srpm from the git clone?