PancakeBot/PancakePainter #40: Mouse icon doesn't change to correct color

Slimmy82, 2015-10-23:
Not sure if this is an issue from the upgrade, but sometimes the color of the curser doesn't change to match the color selected (e.g. stays on darkest even though drawing in the lightest). That being said I have not yet issued the problem of it reverting to the default system curser.

Slimmy82, 2015-10-23:
Scrap the last sentence. Still getting the issue of it reverting to default curser

techninja, 2015-10-23:
Ah well. We'll see if we can't follow up with the Electron Dev Team on this one, though they're likely going to need more information about _when_ it happens and how to reproduce the issue. Document what you can and I can fill in the technical details.

Slimmy82, 2015-10-23:
Will do thank you. I will try and screen video capture the issue.

closed 2018-02-26 by techninja