PancakeBot/PancakePainter #42: An issue someone has with windows.

Slimmy82, 2015-11-25:
I tried PancakePainter in Windows. Ok, but for Windows users I think it is very odd, that we can't specify installation directory. I don't want it on my system disk as it is SSD and so quite small. Please do something about it. Thanks

techninja, 2015-11-25:
The streamlined installer has no such options, and installs in the users `%APPDATA%` folder regardless. We could technically switch to a more old fashioned installer, but this would both diverge us from the community supported install scheme, and require a complete rewrite of the install system and Windows build prep.

Slimmy82, 2015-11-25:
Can you please draft up a simple sentence or short paragraph as to the advantage or way we have it the way we have it to so I have something for the complainers. I appreciate it

Slim Geransar
(646)706-3382
slim@storebound.com<mailto:slim@storebound.com>

[cid:CE92FAD3-CF75-4CDD-A3D3-9B390309CBAB]

techninja, 2015-11-27:
This should work:
> The PancakePainter application utilizes the worlds fastest installer for Windows, by not asking questions most users ignore anyway, and installing into the users home folder to avoid application installation security blocks. While far less adjustable, we provide a better user experience and less friction to a great majority of users.

closed 2015-11-27 by techninja

Slimmy82, 2015-11-27:
Thank you James,

Have a lovely weekend.

Slim Geransar
(646)706-3382
slim@storebound.com<mailto:slim@storebound.com>

[cid:CE92FAD3-CF75-4CDD-A3D3-9B390309CBAB]

Closed #42<https://github.com/PancakeBot/PancakePainter/issues/42>.