PancakeBot/PancakePainter #47: PancakePainter crashes on Mac running 10.8.5

fuzzychocolate, 2016-01-22:
I've downloaded it 3 times, the file size is 132,384,216 bytes, created oct. 22, 2015, version: 0.34.0
Double-click to run and I get: PancakePainter quit unexpectedly. Click Reopen to open the application again. Click Report to see more detailed information and send a report to Apple.
I try that and it occurs again. click report and I get:
Process: Electron [4439]
Path: /Users/USER/Downloads/PancakePainter.app/Contents/MacOS/Electron
Identifier: pancakepainter-main
Version: 0.34.0 (1.0.0-Beta1)
Code Type: X86-64 (Native)
Parent Process: launchd [154]
User ID: 501

Date/Time: 2016-01-21 23:32:26.478 -0800
OS Version: Mac OS X 10.8.5 (12F45)
Report Version: 10

Interval Since Last Report: 75424 sec
Crashes Since Last Report: 35
Per-App Crashes Since Last Report: 3
Anonymous UUID: 0D357569-74D2-42C8-65E6-17F2402844B0

Crashed Thread: 0 Dispatch queue: com.apple.main-thread

Exception Type: EXC_BAD_ACCESS (SIGSEGV)
Exception Codes: KERN_INVALID_ADDRESS at 0x0000000000000000

VM Regions Near 0:

Slimmy82, 2016-01-22:
I have uploaded (If you don’t see anything it maybe still be uploading to dropbox) the version I have onto dropbox. Would you mind trying that? I will send the crash details to the developer. Thank you for the details.

https://www.dropbox.com/sh/51l7bpuqv534huw/AAA1NBwRMnXMs9lCl7jnBqhla?dl=0

Kind Regards,

Slim Geransar
(646)706-3382
slim@storebound.com<mailto:slim@storebound.com>

[cid:CE92FAD3-CF75-4CDD-A3D3-9B390309CBAB]

migpics, 2016-01-22:
Will give it a run and see what comes up with it.

fuzzychocolate, 2016-01-23:
Hi, I downloaded the file from dropbox, unzipped it, there's a contents
directory, with source code, etc. I don't see a compiled program that I can
run....

Slimmy82, 2016-01-23:
Hmm I just uploaded the program. As in a single program file.

fuzzychocolate, 2016-01-23:
I download it and it's saved as a .zip file, I then unzip it and it creates
a 'content' directory with many files.
If I rename it to .app then try to run it, it says:  You can't open the
application "PancakePainter-2.app" because the Classic environment is no
longer supported.

fuzzychocolate, 2016-01-28:
Hey, is anyone looking at this??  I received my Pancakebot today and I can't use it!!!  how frustrating is that!  I can download other people's designs as .pbd, yet you still need painter to convert them to gcode.  Please fix PancakePainter so I can use my Pancakebot........
Can someone send me a homer simpson in gcode that I can print??

migpics, 2016-01-28:
I am.  Let me get some designs to you.  Email me at pancakebot@gmail.com
Miguel

Slimmy82, 2016-01-28:
The designs that show you the result of an actual pancake have all the .gcode files with them.

merkg, 2016-01-29:
It appears the framework used to build the app doesn't really support 10.8.x anymore, so it looks like PancakePainter's minimum requirement is effectively 10.9.

[https://github.com/atom/electron/issues/3530](url)

techninja, 2016-01-29:
Mac 10.7 was the previous breaking release, I guess 10.8 was only a matter of time. Unless the community supports it, we can't have a 10.8 release without knowing what version to build from.. and risk losing APIs and stability along the way. Mountain Lion was released 3 years ago, is it in our interests to keep support for this OS @PancakeBot @Slimmy82? If so there will be quite a bit of work to do to make it happen.

merkg, 2016-01-29:
Meanwhile, I've suggested to the Electron folks that if they're not going to support 10.8, they should at least make it official.

[https://github.com/atom/electron/issues/4273](url)

fuzzychocolate, 2016-01-29:
OK, thanks, I'm able to boot up windows on my Mac and run the Windows
version of pancake painter on my Mac, so now I can print.

techninja, 2018-02-26:
Kind of looks like they've documented this now, and there's no push to improve this direction so going to close the issue for now.

closed 2018-02-26 by techninja