PancakeBot/PancakePainter #50: Line Work Print Order

migpics, 2016-01-22:
The print order of the line work should be simplified.
Currently, the line that is drawn first on the screen, is the last line drawn on the actual pancake.  Any manipulation of the a line, immediately brings it to the top of the stack.
My recommendation is the following.
1.  The drawing of line work on the griddle should emulate the same order that is drawn on the software.  So the first line put down on the software, should be the first line put down on the griddle.
2.  If a line is manipulated, it should not change the order of the stack, but it should stay in the same order it had before it was manipulated.

This serves to ensure the user has control of how the line work is dispensed and the order it is dispensed in.

techninja, 2016-01-22:
To clarify here: Output drawing order was originally based on order drawn, AND line edit reorders. Travel sort was added to make a best guess at most efficient print (as not all users actually think about printing when drawing).

https://github.com/PancakeBot/PancakePainter/blob/master/src/gcode.js#L265-L342

If you remove [the travel sort call](https://github.com/PancakeBot/PancakePainter/blob/master/src/gcode.js#L38-L39), then the layers stay as drawn.

Order of the stack changes can not be done currently through any other method... so you'd need to add support for that before removing it on click.

migpics, 2016-01-22:
The issue seems to arise when a fill is drawn without an outline (assuming the same color code) and as a result, the batter tends to spread beyond the intended limits.  This doesn't happen often but does sometimes.
Could we create the option to choose travel sort prior to outputting the GCode File?

techninja, 2016-01-22:
Yes, among other things. This is also likely where simulator output would be useful.

techninja, 2018-02-26:
Simulator and ordering selection has been managed as part of 1.3.0

closed 2018-02-26 by techninja