PancakeBot/PancakePainter #55: Software Features To Add

Madness44, 2016-01-28:
The software is simple, and i like that. However i feel there are some basic features that need to be added:

Undo/Redo - Others have also mentioned it, nothing worse than deleting an outline you just worked on

Eraser - I know you can select and delete, but a simple eraser to quickly fix those missed freehand outlines. Also allow different sizes/shapes

Simple shapes - add things like circles, stars, ect. that can be easily and evenly made without having to freehand. 

Text - add simple text  so once again it can be somewhat even and alot easier to create.

Basically look at Windows Paint. Add many of those features while still keeping it simple

AstroJetson, 2016-02-03:
Let me add to Madness44 list
-- Text - Need to have the ability to outline text in one color and then fill with another 
-- Shapes -- Need to be able to outline in one color and then fill with another
-- Photo import -- for easy tracing, be able to resize the picture either "on the griddle" or when the import is being done.

migpics, 2016-02-08:
I agree with these all.  Getting the list ready.