PancakeBot/PancakePainter #58: Spirograph drawing

AstroJetson, 2016-02-03:
Part of the fun of the Pancakebot is watching it draw.   My kids are fascinated by Spirograph and the cool shapes that it produces.  

I'd like to suggest a simple Spirograph tool that could be used to create designs.  

From the docs paper.js is based on Scriptographer.  There is already example code on Scriptographer
    http://scriptographer.org/scripts/general-scripts/spirograph/

You can also go crazy and add different base sequences has shown here:
   http://scriptographer.org/scripts/general-scripts/roulettes/

Thanks!

migpics, 2016-02-08:
That is a great idea.
I have worked on a spirograph generator but it is still in the beta phase.
http://www.kreativperspektiv.com/spg/

AstroJetson, 2016-02-08:
Very nice, I like the user interface that should make it easy for children
to play with the designs.

I'll be doing a bake run on Wednesday, I'll let you know how I make out.

Thanks!