PancakeBot/PancakePainter #76: Having trouble running source on Mac 10.11.5

seandoyle, 2016-06-03:
Hi - 

I'm new to electron. I'm able to download and make the app and then run it in electron - but there are no menus. I must be missing a step. Here's what I'm doing:

- git clone https://github.com/PancakeBot/PancakePainter.git
- cd PancakePainter
- npm install --force
- electron src/index.html 

At this point I see the picture of the griddle - but there are no controls. If I try running electron like
electron .
I get 
  App threw an error during load
  Error: Cannot find module 'app'

Node version: v6.2.0
Electron version: v1.2.1

Any suggestions welcomed.

Thanks!

techninja, 2016-06-03:
Your electron version is too new for master. Switch to the [1.2.0](https://github.com/PancakeBot/PancakePainter/tree/1.2.0) branch and follow the new instructions (which locks the version down for each release), then just run `npm start` :smile:

seandoyle, 2016-06-04:
Thanks. That works great!

techninja, 2016-06-04:
Glad to help. FYI 1.2.0 will be the next release version after a short period of beta testing, so feel free and experiment and fork from there :boom:

closed 2016-06-04 by techninja