PancakeBot/PancakePainter #86: Can´t install software for MAC

bzambranog, 2016-10-02:
Hi, I just got the pancakebot yesterday and I keep downloading the software for MAC and it just doesn´t seem to work. The problem reads:

[image]

CAN SOMEONE HELP ME PLEASE! CAN´T WAIT TO USE IT! (Im running on MacOS SIERRA 10.12

ssvenn, 2016-10-18:
To fix this problem, just move the PancakePainter application from the Downloads folder to the Applications folder on your system. 

The error message is due to new Gatekeeper system protection mechanisms in Sierra sandboxing the application in a read-only environment as long as it's sitting in the Downloads folder.

I found some more discussion about it here: https://weblog.rogueamoeba.com/2016/06/29/sierra-and-gatekeeper-path-randomization/

markofzorro, 2016-12-19:
I'm running Sierra 10.12.2, upgraded 12/19/2016 on a Mac Pro late 2013. I got the error when I downloaded the beta version and tried to tun it from Downloads. It leads fine once I loaded it from /Applications.

robnagler, 2017-01-03:
Having the same problem. I'm loading it from /Applications and still getting the error. I cleared ~/Library:

```
rm -rf Application\ Support/PancakePainter Caches/PancakePainter Preferences/pancakepainter-main.plist Saved\ Application\ State/pancakepainter-main.savedState
```

Still no good.

robnagler, 2017-01-03:
Figured it out. You can't install applications with "mv". You have to use Finder so that it informs Gatekeeper correctly.

If you installed with "mv", you'll probably need to remove the exception with "spctl":

```
spctl --remove /Applications/PancakePainter.app
```

You don't need to clear ~/Library after the reinstall of PancakePainter.app.

techninja, 2018-02-26:
I _think_ this is a duplicate, going to mark as resolved/closed for now.

closed 2018-02-26 by techninja