PancakeBot/PancakePainter #90: PancakePainter won't load correctly

Ckhayat, 2017-01-09:
The software program won't load correctly.  I know this is an issue that has been logged before but it doesn't appear to have ever been resolved.  This is the 4th time we have attempted to resolve this in some manner.  If we don't get a response within 24 hours, we will return the pancakebot

Slimmy82, 2017-01-09:
Can you please explain exactly what system you have and what error you're getting. We are finalizing a new version of the software with updated installer, auto trace feature and many more improvements which we plan to release shortly.

Ckhayat, 2017-01-09:
I am currently using windows 8.1 on the surface pro.  The error I've been getting happens after the program downloads.  It comes up and instead of showing the tray in the middle with the tools at the top, it shows two no image icons.

Slimmy82, 2017-01-10:
Thank you for letting us know. I am not too familiar with surface pro, but I will ask the developer to have a look into it for sure. If there are any other details you think that may help us, I would appreciate it. This may include screenshots or video captures.

Ckhayat, 2017-01-10:
Here is a screen shot of what the pancake painter looks like when it opens.