PancakeBot/PancakePainter #101: Error: A JavaScript error occurred in the main process

guruglobal, 2018-01-31:
When I try to run from Mac OSX 10.13.1 I get a popup with the heading:
"A JavaScript error occurred in the main process"

And the dump:

Uncaught Exception:
Error: EROFS: read-only file system, open '/private/var/folders/dc/w7kzcndn5h5gb_sbtyg82rj80000gn/T/AppTranslocation/41CED245-9736-4AA4-B606-C63C4EB6CBC5/d/PancakePainter.app/Contents/Resources/app/settings.json'
    at Error (native)
    at Object.fs.openSync (fs.js:584:18)
    at Object.module.(anonymous function) [as openSync] (ELECTRON_ASAR.js:158:20)
    at Object.fs.writeFileSync (fs.js:1234:33)
    at Object.fsPlus.writeFileSync (/private/var/folders/dc/w7kzcndn5h5gb_sbtyg82rj80000gn/T/AppTranslocation/41CED245-9736-4AA4-B606-C63C4EB6CBC5/d/PancakePainter.app/Contents/Resources/app/node_modules/fs-plus/lib/fs-plus.js:279:17)
    at Object.app.settings.save (/private/var/folders/dc/w7kzcndn5h5gb_sbtyg82rj80000gn/T/AppTranslocation/41CED245-9736-4AA4-B606-C63C4EB6CBC5/d/PancakePainter.app/Contents/Resources/app/src/main.js:77:10)
    at Object.app.settings.load (/private/var/folders/dc/w7kzcndn5h5gb_sbtyg82rj80000gn/T/AppTranslocation/41CED245-9736-4AA4-B606-C63C4EB6CBC5/d/PancakePainter.app/Contents/Resources/app/src/main.js:94:12)
    at settingsInit (/private/var/folders/dc/w7kzcndn5h5gb_sbtyg82rj80000gn/T/AppTranslocation/41CED245-9736-4AA4-B606-C63C4EB6CBC5/d/PancakePainter.app/Contents/Resources/app/src/main.js:98:16)
    at start (/private/var/folders/dc/w7kzcndn5h5gb_sbtyg82rj80000gn/T/AppTranslocation/41CED245-9736-4AA4-B606-C63C4EB6CBC5/d/PancakePainter.app/Contents/Resources/app/src/main.js:33:3)
    at Object.<anonymous> (/private/var/folders/dc/w7kzcndn5h5gb_sbtyg82rj80000gn/T/AppTranslocation/41CED245-9736-4AA4-B606-C63C4EB6CBC5/d/PancakePainter.app/Contents/Resources/app/src/main.js:181:1)

Rfife72, 2018-02-18:
I had the same issue.. I was running it from the desktop and that is the source of the issue.  I was able to solve it by moving the app into the applications folder (imagine that).  Hope this helps!

techninja, 2018-02-26:
Yep, that should do it. Closing the issue.

closed 2018-02-26 by techninja